earlephilhower/ESP8266Audio #589: SPDIF output not working on ESP32

vshymanskyy, 2022-12-03:
Test project:
[Audio_SPDIF_ESP32_fail.zip](https://github.com/earlephilhower/ESP8266Audio/files/10147135/Audio_SPDIF_ESP32_fail.zip)

## ESP8266:
```
pio run -e esp8266 -t upload
pio run -e esp8266 -t uploadfs
```
**Result:** LED is connected to RX,GND. LED light is (visually) stable, audio plays OK.

## ESP32:
```
pio run -e esp32 -t upload
pio run -e esp32 -t uploadfs
```
**Result:** LED is connected to gpio27,GND. LED light is unstable, audio is **not playing**.

pwilkowski, 2023-06-04:
I have exact same problem, it is not a problem of i2s capabilities because this project is able to output audio:
https://github.com/amedes/esp_a2dp_sink_spdif

Altho from bluetooth and i want to have it play from sd card

Edzelf, 2024-02-08:
I had the same problem.  In  AudioOutputSPDIF.cpp I commented out the line "`rtc_clk_apll_enable(1, 28, 8, 5, 0);`" (near line 187).  Now it works correctly.