earlephilhower/ESP8266Audio #592: Add MCLK output for ESP32

DatanoiseTV, 2022-12-23:
```cpp
    i2s_pin_config_t pins = {
        .mck_io_num = 18, // MCLK
        .bck_io_num = bclkPin,
        .ws_io_num = wclkPin,
        .data_out_num = doutPin,
        .data_in_num = I2S_PIN_NO_CHANGE};
    i2s_set_pin((i2s_port_t)portNo, &pins);
```
```cpp
      i2s_config_t i2s_config_dac = {
          .mode = mode,
          .sample_rate = 44100,
          .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
          .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
          .communication_format = comm_fmt,
          .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // lowest interrupt priority
          .dma_buf_count = dma_buf_count,
          .dma_buf_len = 128,
          .use_apll = use_apll, // Use audio PLL
          .tx_desc_auto_clear = true, // Silence on underflow
          .fixed_mclk = 1, // constant MCLK
          .mclk_multiple = I2S_MCLK_MULTIPLE_256, // fs*256
          .bits_per_chan = I2S_BITS_PER_CHAN_DEFAULT // Use bits per sample
      };
```

earlephilhower, 2022-12-23:
Would you like to do a PR?  You may need to add `#ifdef`s around things since the Espressif SDK seems to play with the I2S structs frequently, and the different boards all seem to have different I2S structs/capabilities.

DatanoiseTV, 2022-12-24:
PR: https://github.com/earlephilhower/ESP8266Audio/pull/594

closed 2023-01-19 by DatanoiseTV