earlephilhower/ESP8266Audio #593: ESP32 - SPIFFS was not declared in this scope

ArkasDev, 2022-12-23:
I have already successfully played sounds on the esp8266. Now I wanted to develop the whole thing for the esp32, but the following error message occurs when compiling.

```
In file included from src/main.cpp:2:
.pio/libdeps/esp32dev/ESP8266Audio/src/AudioFileSourceSPIFFS.h: In constructor 'AudioFileSourceSPIFFS::AudioFileSourceSPIFFS()':
.pio/libdeps/esp32dev/ESP8266Audio/src/AudioFileSourceSPIFFS.h:36:49: error: 'SPIFFS' was not declared in this scope
     AudioFileSourceSPIFFS() : AudioFileSourceFS(SPIFFS) { };
                                                 ^~~~~~
.pio/libdeps/esp32dev/ESP8266Audio/src/AudioFileSourceSPIFFS.h:36:49: note: suggested alternative: 'SPI_CS1'
     AudioFileSourceSPIFFS() : AudioFileSourceFS(SPIFFS) { };
                                                 ^~~~~~
                                                 SPI_CS1
.pio/libdeps/esp32dev/ESP8266Audio/src/AudioFileSourceSPIFFS.h: In constructor 'AudioFileSourceSPIFFS::AudioFileSourceSPIFFS(const char*)':
.pio/libdeps/esp32dev/ESP8266Audio/src/AudioFileSourceSPIFFS.h:37:69: error: 'SPIFFS' was not declared in this scope
     AudioFileSourceSPIFFS(const char *filename) : AudioFileSourceFS(SPIFFS, filename) {};
                                                                     ^~~~~~
.pio/libdeps/esp32dev/ESP8266Audio/src/AudioFileSourceSPIFFS.h:37:69: note: suggested alternative: 'SPI_CS1'
     AudioFileSourceSPIFFS(const char *filename) : AudioFileSourceFS(SPIFFS, filename) {};
                                                                     ^~~~~~
                                                                     SPI_CS1
src/main.cpp: In function 'void setup()':
src/main.cpp:13:3: error: 'SPIFFS' was not declared in this scope
   SPIFFS.begin();
   ^~~~~~
src/main.cpp:13:3: note: suggested alternative: 'SPI_CS1'
   SPIFFS.begin();
   ^~~~~~
   SPI_CS1
```

I used a simple example:

**main.cpp**
```
#include <Arduino.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSPIFFS *file;
AudioOutputI2SNoDAC *out;
void setup()
{
  Serial.begin(115200);
  delay(1000);
  SPIFFS.begin();
  file = new AudioFileSourceSPIFFS("/jamonit.mp3");
  out = new AudioOutputI2SNoDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop(); 
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
```

With the follwing settings:

**platformio.ini**
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200

lib_deps = 
    ESP8266Audio
```

inesvk, 2022-12-30:
I have the same issue. Please help! Arduino Version 1.8.19 .

hizbi-github, 2022-12-31:
The error occurs since the SPIFFS library is missing. Please add the following at top of your code (below <Arduino.h>):

#include <SPIFFS.h>

Additionally, you may also need to add the "FS" dependency in the platformio.ini file:

lib_deps = 
	FS

sxy-sun, 2023-04-13:
> The error occurs since the SPIFFS library is missing. Please add the following at top of your code (below <Arduino.h>):
> 
> #include <SPIFFS.h>
> 
> Additionally, you may also need to add the "FS" dependency in the platformio.ini file:
> 
> lib_deps = FS

After adding #include <SPIFFS.h> to my main.cpp as @hizbi-github suggested, I still have the compile error. However after adding it to my AudioFileSourceSPIFFS.h file the problem solved!

DKNS-JCC, 2024-08-14:
> AudioFileSourceSPIFFS.h file the problem solved!

This fixed mine, thanks for help