earlephilhower/ESP8266Audio #601: Running a wav or AAC file more than once.

elmoray1, 2023-02-17:
I am using a Raspberry Pi Pico with a 98357A, and I am getting a strange issue. Once any file plays once and gets completed.  I delete the file, generator, and I2S output.   I then start over to play the same or different audio, and the isRunning stays true with no audio output.  Reboot the Pico, and it again will only play one time.   Unclear what I am doing wrong, but somehow the i2c/Dac is left in a bad state.  I would try to give more info, but I am having issues getting serial data from the audioLogger connection. 

Here is one example: 

`#include <Arduino.h>
#include "AudioGeneratorAAC.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourcePROGMEM.h"
#include "sampleaac.h"

AudioFileSourcePROGMEM *in;
AudioGeneratorAAC *aac;
AudioOutputI2S *out;

void setup()
{
  Serial.begin(115200);

  audioLogger = &Serial;

}

void loop()
{

  in = new AudioFileSourcePROGMEM(sampleaac, sizeof(sampleaac));
  aac = new AudioGeneratorAAC();
  out = new AudioOutputI2S();
  out->SetPinout(3,4,2);

  aac->begin(in, out);

  while (aac->isRunning()){
  if (aac->isRunning()) {
    aac->loop();
  } else {
    Serial.printf("AAC done\n");
    delay(1000);
  }
  }

  delete in;
  delete aac;
  delete out;

}`

elmoray1, 2023-02-18:
Another data point for you.  I used the same code above for the ESP32, and it had no issues playing the sound multiple times. The problem is more in the i2s raspberry pi pico driver.  It may have never got seen because all the example code that none of the sample code runs a file more than once.  So the problem has to do with the close of the i2s.  To get it unstuck, a simple reload of the code will get it to play again once without a hard reboot. Whatever code initializes the i2s after reset holds the key.

LinusHeu, 2023-06-12:
Maybe this fixes your issue?
https://github.com/earlephilhower/arduino-pico/pull/1526

nils-trubkin, 2023-06-27:
I have almost an identical issue.

I'm trying to play a WAV file on my Qt Py RP2040 through an I2S DAC (98357A) from an SD card. The problem is that every second time I try to play the file, the begin() function returns true, but no sound is produced. According to the logs, the sound should have stopped playing once completed but continues indefinitely (with no actual sound coming). The loop() of the WAV generator will continue returning true indefinitely.

I suspect the issue is related to the i2s initialization/deinitialization in either this library or the abovementioned one. I've tried different methods to resolve the issue, including reinitializing the I2S, sd source, and WAV generator, but the only solution that works is restarting the entire board. Do you have any suggestions on how I can fix this problem? I'd appreciate any advice on playing multiple sounds separately without repeatedly restarting the entire board. Thank you.