earlephilhower/ESP8266Audio #623: ESP32-S3 Support?

builderdev212, 2023-05-08:
I was wondering if there was any plans to support the ESP32-S3 or if anyone has gotten it to work properly? My setup for the ESP32 works fine, however there is very little documentation for the proper pins for I2S when it comes to the ESP32-S3. Any help/advice would be very much appreciated. Thanks in advance!

eokgnah, 2023-05-19:
pls add the esp32-s2 as well.
the "normal" pins for an external dac are not on the board.
the io's are missing.

lyusupov, 2023-05-20:
https://github.com/lyusupov/ESP8266Audio/commit/53802bd21fcfe3ead516cdcaf3ac0b582d55ff51

trailhead, 2023-10-13:
It works.  The pins can be whatever you want.  For example I'm using:
`  out->SetPinout(39, 40, 38);
`