earlephilhower/ESP8266Audio #628: [Feature request] HTTPS support

rebane2001, 2023-05-18:
At the moment this library can only play content from insecure HTTP sources, but it'd be awesome to see it play HTTPS as well since HTTPS-only sources are becoming more and more common.

I got HTTPS to work for myself by using `WiFiClientSecure` in `AudioFileSourceHTTPStream.h` and setting the right root ca (or you can do `client.setInsecure()`) in `AudioFileSourceHTTPStream.cpp`. I imagine this isn't the best way to go about things, hence the issue and not a PR, but it works great for me.

Perhaps it could also make sense to instead let the user provide their own WiFiClient along the lines of `file = new AudioFileSourceHTTPStream(URL, client);` so that the user can use their own WiFiClientSecure and root ca instead of delegating this to the audio library.

positron96, 2023-09-01:
Huh, so that's why I couldn't play https streams...

3ton, 2024-08-26:
> I got HTTPS to work for myself by using `WiFiClientSecure` in `AudioFileSourceHTTPStream.h` and setting the right root ca (or you can do `client.setInsecure()`) in `AudioFileSourceHTTPStream.cpp`. I [...]

This not work for me (ESP8266Audio version 1.9.7)

closed 2025-10-23 by github-actions[bot]