earlephilhower/ESP8266Audio #649: Compile error when using AudioOutputULP.h

CobaltEcho, 2023-09-08:
When I use the AudioOutputULP.h library, I get an error on Build/Compile:

```undefined reference to `vtable for AudioOutputULP'```

Full message:
```
c:/users/echo/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `vtable for AudioOutputULP'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

It compiles fine using the AudioOutputI2S library (using internal DAC) `output = new AudioOutputI2S(0, 1);`, but since I'm using the internal DAC, I would like to free up my I2S and also attempt to prevent popping as noted [here in issue #406 ](https://github.com/earlephilhower/ESP8266Audio/issues/406#issuecomment-952642137).

ben25belot, 2024-03-18:
Hi,

Yes, I'm also trying to eliminate the popping sounds.
I managed to do it with this OutputULP by removing line 21 of the file AudioOutputULP.cpp (we could also modify it for better protection).
`#if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3`

Everything works for me, without any popping sounds!

CobaltEcho, 2024-10-16:
I got it to work using platformIO with:

```
platform = espressif32 @ ^6.4.0
board = esp32dev
framework = arduino, espidf
lib_deps = 
	earlephilhower/ESP8266Audio@^1.9.7
**_build_flags = -DCONFIG_IDF_TARGET_ESP32_**
```