earlephilhower/ESP8266Audio #654: Any exaples for RP2040?

Stass48, 2023-11-25:
Hello! I want to play MP3 from SD Card on RP2040. Can someone help me with it? Examples in library is not working for me.

maxgerhardt, 2023-12-05:
Well now this is linked to https://github.com/earlephilhower/arduino-pico/discussions/1854

Stass48, 2024-03-05:
Here's the working code. You need to solder a jumper on the PCM5102 module (shown in the photo) and power the module from 5 V. To avoid jerks during playback, you need to increase the SPI speed, as shown in the code. Also, use overclocking up to 200 MHz. Under these conditions, MP3 320 kbps Stereo is played.

[image]

```
#include "AudioFileSourceSD.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioOutputI2S *out;

// SPI0 for internal SD card
const int INT_SD_SCK = 18;
const int INT_SD_MOSI = 19;
const int INT_SD_MISO = 20;
const int INT_SD_CS = 21;

// You may need a fast SD card. Set this as high as it will work (40MHz max).
#define SPI_SPEED SD_SCK_MHZ(40) // Why not 50?

// Pins for I2S
const int PIN_BCLK = 2; // BCK
const int PIN_WCLK = 3; // LCK
const int PIN_DOUT = 4; // DIN
// Solder SCK jumper on module! VCC of PCM5102 module = 5V!

void setup() {
  SPI.setRX(INT_SD_MISO);
  SPI.setTX(INT_SD_MOSI);
  SPI.setSCK(INT_SD_SCK);
  SPI.setCS(INT_SD_CS);

  Serial.begin(115200);
  delay(1000);

  if (SD.begin(INT_SD_CS, SPI_SPEED, SPI)) Serial.println("SD OK!");
  Serial.println("Sample MP3 playback begins...");

  audioLogger = &Serial;
  file = new AudioFileSourceSD("/demo.mp3");
  out = new AudioOutputI2S();
  out->SetPinout(PIN_BCLK, PIN_WCLK, PIN_DOUT);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop() {
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.println("MP3 done");
    delay(1000);
  }
}
```

Stass48, 2024-03-05:
Now I am interested in other questions.
In what mode does the memory card work? DEDICATED_SPI or SHARED_SPI? How can you control these modes? For example, set the DEDICATED_SPI mode.
If I have two SD cards, how can I choose which card will play files from?
For example, if I write:

```
#define SPI_CLOCK SD_SCK_MHZ(50)
SdFs rdSd;
FsFile rdFile;
#define SD_RD_CS_PIN 17
#define SD_RD_CONFIG SdSpiConfig(SD_RD_CS_PIN, DEDICATED_SPI, SPI_CLOCK, &SPI)

SdFs wrSd;
FsFile wrFile;
#define SD_WR_CS_PIN 13
#define SD_WR_CONFIG SdSpiConfig(SD_WR_CS_PIN, DEDICATED_SPI, SPI_CLOCK, &SPI1)

...

if (!rdSd.begin(SD_RD_CONFIG)) rdSd.initErrorHalt("rdSd");
if (!wrSd.begin(SD_WR_CONFIG)) wrSd.initErrorHalt("wrSd");
```

**How do I play a file from each memory card?

Another interesting question. Does the library support multithreading? RP2040 has 2 cores. Perhaps by using both cores, overclocking could be avoided.**

galinawind, 2024-09-29:
How did you get this to work exactly? When I try the example (while giving my value for the SDCS pin) the whole thing just hangs on the SPI.setCS() call. And I mean hangs, the whole thing stops responding. My hardware is a Feather RP2040 connected to MAX98357 amplifier. My SD card reader is on my ST7789 display unit, but that shouldn't cause anything to hang up as the SD card has its own wiring and goes to its own pins that are not shared by anything else.

Stass48, 2024-10-04:
> How did you get this to work exactly? When I try the example (while giving my value for the SDCS pin) the whole thing just hangs on the SPI.setCS() call. And I mean hangs, the whole thing stops [...]

Miracles don't happen. Everything works for me. I wouldn't give you non-working code. If you have some freezes, then look for the reason somewhere else. Have you tried uploading only my code to a clean sketch in your hardware? Will it freeze?