earlephilhower/ESP8266Audio #672: just noise on Wemos LOLIN D32

sartyx, 2024-03-20:
Hi,

I just tried the PlayWAVFromPROGMEM example on a LOLIN Wemos D32.

Though I am getting only noise. 
I started with using the pins:

25,33,26 = LRC,BCLK,DOUT

25,26 are DAC pins on the D32 board, 

I used them in an alternative way with the XT-DAC lib and a PAM amp, so I left the wiring where it was but got only loud noise with the ESP8266Audio lib, the XT-I2S at least plays some sound but it is also chopped and not clean as it should be.  

Then I thought I might use some different pins and switched over to 

  bclkPin = 22;
  wclkPin = 21;
  doutPin = 13;

( I hardcoded the pins to the lib, cause I found no description how to change them otherwise. ;-)  )

This leads to a slightly better result but still the sound is more noise than anything else. Far far away from what it should sound like
which would be this http://ccrma.stanford.edu/~jos/wav/viola.wav

MAX98357A board for the I2S sound used
ESP8266Audio version=1.9.7

Anyone with more luck?

earlephilhower, 2024-03-21:
That example [doesn't use an I2S DAC](https://github.com/earlephilhower/ESP8266Audio/blob/0abcf71012f6128d52a6bcd155ed1404d6cc6dcd/examples/PlayWAVFromPROGMEM/PlayWAVFromPROGMEM.ino#L12C1-L12C20), just a delta-sigma R/C filter on the data pin.  Change to an `AudioOutputI2S` if you have a real DAC attacked.

closed 2024-03-21 by earlephilhower

sartyx, 2024-03-21:
excellent, thanks - me stupid ;-)