earlephilhower/ESP8266Audio #673: how do i modify the Output Pin for NoDAC

Buttermesserente, 2024-03-24:
So i built the example with a single transistor as a amplifier and it uses the TX pin connected to the base to control the frequencies. for my project it would be important to use some other pin as the output source. is there any way to do this?

kleinesfilmroellchen, 2024-03-29:
On the ESP8266, the I2S data pin is hard wired to the same GPIO as TX. Just like the pins for SPI1, SPI2 and I2C are hardwired too. The I2S data pin is required for the ΔΣ-DAC in the "NoDAC" output generator. You can't switch the output pin for other I2S-based outputs (DAC or SPDIFF) either.

xXxOlivierxXx, 2024-08-14:
Add the following to your code:

#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_4 // BCLK
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_5 // WSEL
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_17       // pin 17 on the ESP32S2 where the speaker is connected, or replace with any other pin