earlephilhower/ESP8266Audio #683: ESP8266Audio compilation error in V3.0.0-Release of Arduino-esp32

Zhentao-Lin, 2024-05-31:
这是一个非常实用且好用的库，我非常喜欢使用它来做一些音频方面的开发。
今天我在更新我的Arduino-esp32到3.0.0-Release版本后，发现编译失败了。由于我个人能力水平有限，我向您请求帮助。下面是我的报错信息。

[image]

再次谢谢您。

earlephilhower, 2024-05-31:
Espressif seems to have broken existing code with the (IMHO reasonable and a good idea in the long term!) change from `WiFiClient` to `NetworkClient` in 3.0.0.  Since they have an entire Ethernet MAC onboard in addition to the WiFi radio, it makes sense.  On the Pico we ended up keeping `WiFiClient` and just letting it work on anything, wired or wireless.

All network code for the ESP32 will need to be rewritten to use the new class names.  I'm not really an ESP32 user so hopefully someone here will have the time and knowledge to to the transition and throw a PR out...

Lebo77, 2024-06-06:
If I don't use the WiFi functions in this library (all the audio I am playing is local) is there a workaround to comment out all the network stuff?

PR #685 (2024-06-06): Update for ESP32 Arduino 3.x
Implement the renamed network client and other API changes when built under the 3.x branch of the Arduino ESP32 core.

Disable building TSF on the ESP32, the Xtensa G++ backend bug is still there.

Minimal changes to get I2S to build under new IDF.  Deprecated but not removed, so warnings are expected but should not be fatal in use.

Fixes #683
commit a6141473 (2024-06-06): Update for ESP32 Arduino 3.x (#685)

Implement the renamed network client and other API changes when built
under the 3.x branch of the Arduino ESP32 core.

Disable building TSF on the ESP32, the Xtensa G++ backend bug is still there.

Minimal changes to get I2S to build under new IDF.  Deprecated but
not removed, so warnings are expected but should not be fatal in
use.

Fixes #683
closed 2024-06-06 by earlephilhower