earlephilhower/ESP8266Audio #692: library.properties uncorrected resulting in Arduino lib display version 1.9.7

Zhentao-Lin, 2024-06-29:
name=ESP8266Audio
version=1.9.8  **(This should correspond to tag)**
author=Earle F. Philhower, III
maintainer=Earle F. Philhower, III
sentence=Audio file and I2S sound playing routines for ESP8266, ESP32, and Raspberry Pi Pico RP2040
paragraph=Decode compressed MP3, AAC, FLAC, Screamtracker MOD, MIDI, RTTL, TI Talkie, and WAV and play on an I2S DAC or a software-driven delta-sigma DAC and 1-transistor amplifier.
category=Signal Input/Output
url=https://github.com/earlephilhower/ESP8266Audio
architectures=esp8266,esp32,rp2040

egnor, 2024-08-07:
Yes! @earlephilhower this [prevents the Arduino library indexer from actually picking up 1.9.8](https://downloads.arduino.cc/libraries/logs/github.com/earlephilhower/ESP8266Audio/) (which is quite unfortunate because 1.9.8 is needed for arduino-esp32 3.x support!):

```
2024/08/07 03:20:08 Checking out tag: 1.9.8
2024/08/07 03:20:08 Release ESP8266Audio:1.9.7 already loaded, skipping
```

Fix should be trivial -- properly update `library.properties` to 1.9.9 and then tag 1.9.9.

Zhentao-Lin, 2024-08-07:
Yes, I'm just giving you feedback so we can correct it next time.

egnor, 2024-08-07:
I'm not a maintainer, just another kibitzer agreeing with you! Leave it open until @earlephilhower or another maintainer acknowledges it?

closed 2024-09-22 by earlephilhower