earlephilhower/ESP8266Audio #697: analogRead when using ESP8266Audio

sittelle, 2024-08-09:
Hello,

I successfully tested the example "PlayAACFromPROGMEM". When I add an analogRead (any pin), my ESP32-WROOM-32U gets stuck in a boot loop. I guess that this might not be a library problem but maybe someone here can help me.
As I wrote, I have changed nothing in the example code but added a "analogRead(XX)" to it.

Best regards

sittelle, 2024-08-12:
This library seems to be working with Arduino core ESP32 version 2.x, but not with 3.x.

dired, 2024-08-14:
same here. It is something with drivers that got updated to be more "smart", which I found out so far by searching for the error message: 

`E (154) ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver`

sittelle, 2024-08-15:
Hi @dired,
downgrading to ESP32 core 2.x happens to solve this issue. As you said, it has something to do with the new driver for core 3.x.
Regards

dired, 2024-08-18:
Because of this error I finally tried platformio (glad about it)

There I was able to change the esp-idf version via `platformio.ini`:
```
platform = espressif32@5.3.0
```
with [5.3.0 being the latest version with the 4.0 version](https://github.com/platformio/platform-espressif32/releases/tag/v5.3.0) (which (new drivers in 5, works in "4") is what i read in my googling). 

That worked, but then out of curiosity I tried with the latest release (removing "@5.3.0"). And except for a warning, it compiles and works fine.

So my guess is that it was fixed in the latest release 3 weeks ago, but arduino ide bases on an earlier version which gives the error, platformio doesnt.