earlephilhower/ESP8266Audio #698: Platformio darwin_arm64 package not found (Apple M2)

Mir1001, 2024-08-13:
If I upgrade from 1.9.7 to 1.9.8 i get an error:
 ``UnknownPackageError: Could not find the package with 'earlephilhower/ESP8266Audio @ ^1.9.8' requirements for your system 'darwin_arm64'``

```bash
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
build_flags = -D CORE_DEBUG_LEVEL=0 -fmerge-all-constants -D_BSD_SOURCE 
lib_deps = 
	fastled/FastLED@^3.5.0
	earlephilhower/ESP8266Audio@^1.9.8. ;1.9.7 WORKS fine
	akj7/TM1637 Driver@^2.1.2
monitor_speed = 115200
upload_speed = 1000000
```

dired, 2024-08-15:
it's not up-to-date in platformio for 2 years

Mir1001, 2024-08-18:
Platformio can attract new users. People search there also.

dired, 2024-08-18:
true.. maybe someone can take the effort to write and push request the git-pipeline that would automatically update it there. maybe someone can assume the role of caring for only updating the packages over there. I understand it's enough effort already and stuff like this can take out the fun which I personally don't wish for any open-source author.
(I personally though don't even want to check two places if I know one way with basically the same effort always gives me the latest release.)

closed 2024-10-15 by earlephilhower