earlephilhower/ESP8266Audio #719: Playing from internal Flash via LittleFS

RobS2332, 2024-10-30:
HI, I've been messing around with the PlayMIDIFromLittleFS example and the PlayMP3FromSPIFFS example for a couple of days now and don't seem to be getting anywhere.

Seeing as no one has posted an example on how to play a simple mp3 sound effect stored in flash, is it safe to say this is problematic, at least, and I should abandon all hope?!?

ahmadfathan, 2024-11-15:
You can play around with AudioFileSourceBuffer and AudioFileSourcePROGMEM. Basically you're only required to load the bytes from SPIFFS or LittleFS, then pass it to AudioFileSourceBuffer. The rest will be depends on what type of audio you want to play.

RobS2332, 2024-11-15:
Yeah, sorry. I chopped and changed my code a lot. It seems it's the bigger picture of AudioOutputI2S_NoDAC_  and the latest ESP32 revisions, AudioOutputI2S works fine. I've changed this for WAVs and it works from an .h file. I'll start looking again at LittleFS next for storing small sound effect files. Thanks