earlephilhower/ESP8266Audio #721: Build failure with version 1.9.9 for ESP32-C3 target

TheJKM, 2024-12-15:
I updated ESP8266Audio in my project from 1.9.7 to 1.9.9 and since then, I receive the following errors when trying to compile for the ESP32-C3 target platform:

```
    ~/Documents/GitLab/radio-2/firmware    main !2 ?1  pio run --target upload -e esp32-c3-devkit                                                                                                                                                                                                    1 ✘  4s  
Processing esp32-c3-devkit (platform: espressif32; board: esp32-c3-devkitm-1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Removing unused dependencies...
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing ESP8266audio @ ^1.9.9
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESP8266Audio@1.9.9 has been installed!
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266Audio @ 1.9.9
|-- Wire @ 2.0.0
|-- Adafruit GFX Library @ 1.11.11
|-- Adafruit BusIO @ 1.16.2
|-- Adafruit SSD1306 @ 2.5.13
|-- RotaryEncoder @ 1.5.3
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/esp32-c3-devkit/src/AudioPlayer.cpp.o
Compiling .pio/build/esp32-c3-devkit/src/DisplayRenderer.cpp.o
Compiling .pio/build/esp32-c3-devkit/src/RotaryInput.cpp.o
Compiling .pio/build/esp32-c3-devkit/src/StateMachine.cpp.o
Compiling .pio/build/esp32-c3-devkit/src/WifiManager.cpp.o
Compiling .pio/build/esp32-c3-devkit/src/main.cpp.o
Building .pio/build/esp32-c3-devkit/bootloader.bin
Generating partitions .pio/build/esp32-c3-devkit/partitions.bin
esptool.py v4.5.1
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
Compiling .pio/build/esp32-c3-devkit/lib799/FS/FS.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib799/FS/vfs_api.cpp.o
Compiling .pio/build/esp32-c3-devkit/libcd4/SPIFFS/SPIFFS.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib6a0/I2S/I2S.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiAP.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiClient.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiMulti.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiScan.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiServer.cpp.o
Compiling .pio/build/esp32-c3-devkit/lib10e/WiFi/WiFiUdp.cpp.o
Archiving .pio/build/esp32-c3-devkit/libcd4/libSPIFFS.a
Indexing .pio/build/esp32-c3-devkit/libcd4/libSPIFFS.a
Compiling .pio/build/esp32-c3-devkit/lib7e2/WiFiClientSecure/WiFiClientSecure.cpp.o
Archiving .pio/build/esp32-c3-devkit/lib799/libFS.a
Indexing .pio/build/esp32-c3-devkit/lib799/libFS.a
Compiling .pio/build/esp32-c3-devkit/lib7e2/WiFiClientSecure/esp_crt_bundle.c.o
Compiling .pio/build/esp32-c3-devkit/lib7e2/WiFiClientSecure/ssl_client.cpp.o
Archiving .pio/build/esp32-c3-devkit/lib6a0/libI2S.a
Indexing .pio/build/esp32-c3-devkit/lib6a0/libI2S.a
[... 198 lines cut ...]
            ^~~~~~~~~~                                                                        
.pio/libdeps/esp32-c3-devkit/ESP8266Audio/src/AudioGeneratorMIDI.cpp:640:6: error: invalid use of incomplete type 'struct tsf_stream'
   afs->close = &afs_close;                                                                                                          
      ^~                   
In file included from .pio/libdeps/esp32-c3-devkit/ESP8266Audio/src/AudioGeneratorMIDI.cpp:59:                                       
.pio/libdeps/esp32-c3-devkit/ESP8266Audio/src/AudioGeneratorMIDI.h:57:12: note: forward declaration of 'struct tsf_stream'
     struct tsf_stream buffer;                                                                                            
            ^~~~~~~~~~                                                                        
.pio/libdeps/esp32-c3-devkit/ESP8266Audio/src/AudioGeneratorMIDI.cpp:641:6: error: invalid use of incomplete type 'struct tsf_stream'
   afs->size = &afs_size;                                                                                                            
      ^~                 
In file included from .pio/libdeps/esp32-c3-devkit/ESP8266Audio/src/AudioGeneratorMIDI.cpp:59:                                       
.pio/libdeps/esp32-c3-devkit/ESP8266Audio/src/AudioGeneratorMIDI.h:57:12: note: forward declaration of 'struct tsf_stream'
     struct tsf_stream buffer;                                                                                            
            ^~~~~~~~~~                                                                        
*** [.pio/build/esp32-c3-devkit/libf0a/ESP8266Audio/AudioGeneratorMIDI.cpp.o] Error 1                                     
============================================================================================================================================= [FAILED] Took 8.83 seconds ============================================================================================================================================= 
 
Environment      Status    Duration
---------------  --------  ------------
esp32-c3-devkit  FAILED    00:00:08.829
======================================================================================================================================= 1 failed, 0 succeeded in 00:00:08.829 =======================================================================================================================================
```

Error happens on Fedora 41 and macOS 15.2 (Intel), so I assume the build platform is not relevant.

earlephilhower, 2024-12-19:
The ESP32 compilers, at least Xtensa ones, had an internal error that caused them to crash on building TSF.h, so MIDI is not supported on them.  I know they have some RISC-V chips now, so that may need to be re-examined (i.e. only disable on Espressif Xtensa).  The ESP8266 Xtensa compiler works but it's not what Espressif is using.

closed 2024-12-19 by earlephilhower

TheJKM, 2024-12-19:
Did you maybe get my issue wrong? It does NOT compile on RISC-V, however it DOES on Xtensa. The exact same code worked with 1.9.7, so this is a regression.
I'm neither using, nor importing any MIDI related stuff.

earlephilhower, 2024-12-19:
Ah, that makes things more clear.  I can't keep track of their chips and architectures.

In any case, this was because TSF.h was blocked with plain !ESP32, matching your RISCV arch, but the midi generator..CPP  was blocked only on ESP32 and XTENSA.  So it was compiling the object, but not the TSF engine.  The latest git master should clear that up.

TheJKM, 2024-12-19:
Compilation works again with git master. Thank you for the quick fix!

PR #722 (2024-12-19): Only block tsf.h on Xtensa ESP32, not RISCV
Fixes #721
commit cdd07c92 (2024-12-19): Only block tsf.h on Xtensa ESP32, not RISCV (#722)

Fixes #721