earlephilhower/ESP8266Audio #728: Cannot decode BBC AAC stream

martinbrennan, 2025-01-20:
I am using the decoder to succefully decode m4a files (from iTunes) and some internet radio stations - but the decoder generates errors - typically ERR_AAC_NCHANS_TOO_HIGH when I try to decode data from the BBC.
I have tried data delivered with DASH and HLS protocols.
In both these protocols the stream is broken into segments of a few seconds duration.
The streams seem to contain ADTS headers at regular intervals.
I have attached a file containing a segment. (You will have to unzip). It does not start with a header but that is not the problem.
Is BBC using a version of AAC that is not compatible with Libhelix?
Is there something I can do to allow libhelix to decode this stream?
Martin

[bbc.zip](https://github.com/user-attachments/files/18475202/bbc.zip)

earlephilhower, 2025-01-20:
Libhelix here is configured using the default 2 channel max.  If Aunty Beeb is streaming 2.1 or 5.2 that would be an issue and not supported here.  Do you have the exact URL for the stream?  `mediainfo` seems to think the attached sample is only 2.0 but maybe there are occasional LFE or something?

AAC does has frame header detection as part of the normal process (more complicated than MP3, but still always done before a decode attempt) so I believe TS files will run fine.

The only place those errors can be returned are
https://github.com/earlephilhower/ESP8266Audio/blob/3044568f85224e0f5f13b19c15640ef3f1629ff7/src/libhelix-aac/aacdec.c#L358-L360
and
https://github.com/earlephilhower/ESP8266Audio/blob/3044568f85224e0f5f13b19c15640ef3f1629ff7/src/libhelix-aac/aacdec.c#L380-L382

You could add a simple debug output to report what the actual raw stream is decoding in those 2 spots.

martinbrennan, 2025-01-21:
Hello Earl
Thank you for your very rapid reply - very commendable
In fact I have made some progress and planned to answer my own question once I confirm whats happening.
In fact the ts format is a wrapper that contains the aac data as payload - which is why I can see ADTS headers in the data.
The ts format can carry multiple streams so its not quite as simple as looking for a tag and then extracting the following aac data. Had that been the case libhelix might have decoded it.
In fact the aac data gets chopped up so the ts stream must be parsed and the aac data re-assembled before delivering to libhelix.
Martin

closed 2025-01-21 by martinbrennan