earlephilhower/ESP8266Audio #730: PlayWAVFromFunction example produces loud garbled noise instead of tone

mujido, 2025-02-18:
I'm experimenting with a I2S breakout board from SparkFun using the MAX98357A chip from an ESP32-WROOM-32 (original ESP32).

I was able to run the PlayAACFromPROGMEM example without any problems. Maybe a slight distortion but I think that's a gain issue. I tried switching to the PlayWAVFromFunction example because I wanted to generated a pure tone for testing. Instead this example plays very loud and painful noises. There is some kind of repetition in the sound so it does seem to be periodic but I'm not sure the exact problem.

I'm trying to debug the code and see the difference between the two examples.

Also, what can you explain what the fmod() line is doing here? I don't understand what it means by trying to linearize the signal.

```cpp
  file->addAudioGenerators([&](const float time) {
    float v = sin(TWO_PI * hz * time);  // generate sine wave
    v *= fmod(time, 1.f);               // change linear
    v *= 0.5;                           // scale
    return v;
  });

```

mujido, 2025-02-18:
It seems it might be related to WAV functionality in general. The PlayWAVFromPROGMEM example produces similar output. Other examples I've been able to try (non-WAV) have worked fine.

alastaira, 2025-05-12:
That's because the WAV examples use the `AudioOutputI2SNoDAC` sink.

If you're using a MAX98357A module, you want `AudioOutputI2S` instead....