earlephilhower/ESP8266Audio #746: Consider re- or dual-licensing the library under LGPL

bandrews, 2025-05-13:
Currently, the ESP8266Audio library is licensed under the GPL 3.0 license.  One of the features of this license is that it requires all code it is linked with to _also_ be licensed under GPL 3.0, which means this library can't be used for anything that links with any commercial code.

I'm working on an ESP32-based device that, as one of several features, will offer simple audio playback via an I2S audio chip.  Audio is one of many features of the device, and it depends on some commercial device driver libraries from vendors which are closed-source.  (Ugh.)  But I'd love to adopt ESP8266Audio rather than falling back on the ESP-IDF.

Would you consider re- or dual-licensing this library under the LGPL?  This would maintain the requirement for all contributions to be shared openly, but would permit linking against closed-source code.  In addition to resolving theoretical conflicts with closed-source libraries, it would also encourage use of ESP8266Audio instead of the ESP-IDF audio bits and bobs in commercial projects like mine overall, increasing the likelihood of community bug fixes and contributions that benefit all users of the library.

(Totally understand if the answer is no due to a principled belief in the full GPL to explicitly exclude uses like mine, but figured it was worth asking!)

earlephilhower, 2025-05-14:
I don't really mind for my own code (the majority of which is plumbing and memory + fixed-point math optimization and probably has a few contenders for [The Daily WTF](https://thedailywtf.com/)), but the really interesting bits of this library are licensed under other terms that I can't change.

For example, the MP3 decoder is [GPL](https://github.com/earlephilhower/ESP8266Audio/blob/master/src/libmad/LICENSE), and the AAC is [RealNetworks](https://github.com/earlephilhower/ESP8266Audio/blob/9dee3010c1c6acd6603523ead5a38284cf451127/src/libhelix-aac/aacdec.c#L1-L36).

Stripping those bits out would leave a much less useful library IMHO...

bandrews, 2025-05-14:
You undersell the high value of that plumbing - and of the sensible and easy-to-integrate library design overall!  You've done great work with this project!

That rationale makes a lot of sense, though - I hadn't considered the upstream dependencies.  For my selfish-and-specific purpose, WAV and OGG playback would actually be 'enough'... but I also agree that the support for MP3 and AAC are critical to the library's value to most others.

I think a prerequisite for any potential dual-license subset would need to be preserving the ability to use the full version those in projects where the GPL is sufficient (which is probably most folks!), and I'm not really sure if that's worth the effort (or even possible at all?)...

Thanks for the response!