earlephilhower/ESP8266Audio #748: Seems to crash or conflict with ESP32 Arduino Core 3.2.0

UnexpectedMaker, 2025-05-15:
Works fine in 3.1.1, 3.1.2 and 3.1.3, but in 3.2..0 it crashes.

```
Users/seon/Library/Arduino15/packages/esp32/tools/esp-x32/2411/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/seon/Library/Caches/arduino/sketches/A500AD086F31F98F43568D72CBF7FDE5/libraries/ESP8266Audio/objs.a: member /Users/seon/Library/Caches/arduino/sketches/A500AD086F31F98F43568D72CBF7FDE5/libraries/ESP8266Audio/objs.a(AudioFileSourceBuffer.cpp.o) in archive is not an object
```

earlephilhower, 2025-05-15:
I can't reproduce that on my own install of 3.2.0.  What example are you compiling and what board?

The error shown looks like you ran out of disk space, AV went wild, or there was a GCC error, resulting in the compiled object file not existing or being invalid, and unrelated to the library...

UnexpectedMaker, 2025-05-15:
ESP32S3, my own board, bur currently set to ESP32S3 Dev Module -  and my own code that works fine in previous core versions, and in Arduiuno I can't compile - it keeps throwing that error, and it might be a red herring, but in platformIO, it hard crashes when I try to play audio using pioarduino 3.2.0 and works as expected in 3.1.3.  If I dont tell it to play audio, no crash.

I have it play a wav from progmem on bootup.

repoed on 2 different Macs, runbning macOS 14 and 15, and just had someone repo on a Windows 11 PC in platformIO on same code..

UnexpectedMaker, 2025-05-15:
I'll try to find some time on the weekend to make a simple repo cas for you.

earlephilhower, 2025-05-15:
I just set the IDE to that Xtensa-based ESP32S3 dev module and built the StreamMP3FromHTTP example which uses the object mentioned in that error and it built the ELF fine for me.  My earlier test was R5 and that also was good.

I suggest enabling the full compile logs and see what `g++` command line is building that `.o` and seeing if your computer is throwing an error on it.  As I said, that's not a source code error, it's the object file that GCC generates being corrupted in some way.

You could also try exiting the IDE completely and deleting the cached files in `/Users/seon/Library/Caches/arduino/` which maybe got corrupted and not regenerated for you.